bernhard-42/vscode-ocp-cad-viewer #136: Object is not centered properly when tools are disabled.

roipoussiere, 2024-12-17:
With the glass mode enabled (default), the object is shifted on the right for a better visualization of the tree.
But when `tools` option is set to false, there is no tree, and so the object appears not centered.

This bug is not present when glass mode is disabled.

Example:

```py
import build123d as _
import ocp_vscode as ov

box = _.Box(3, 3, 1)
ov.show_object(box, tools=False, clear=True)
```

Displays:

[image]

bernhard-42, 2025-02-06:
Cannot reproduce it. Is perfectly centered on my machine:

[image]

bernhard-42, 2025-02-06:
There was a bug. Not exactly as you described it, but the some combinations of `glass` and `tools` showed wrong results. 
Will be fixed in the next release

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42