bernhard-42/vscode-ocp-cad-viewer #139: reset_camera=Camera.CENTER doesn't center

voneiden, 2024-12-29:
Python: 3.12
ocp_vscode: 2.6.1

Running on Wayland.

```python
from build123d import Box, BuildPart
from ocp_vscode import Camera, show_all

with BuildPart() as part_builder:
    Box(10, 10, 10)

show_all(reset_camera=Camera.CENTER)
```
When running the code twice, I get the following output with the box half clipped outside of the view

[image]

I poked around a bit and my first impression is that if I move this line https://github.com/bernhard-42/vscode-ocp-cad-viewer/blob/f7df751e918762186f281b5cb06d00807f8a7b21/resources/viewer.html#L509 up above before `p` is defined the box will end up in a better position, but it's still not quite centered:
[image]

I've tested some previous versions and looks like in 2.3.3 it works as expected and the box is perfectly centered.

bernhard-42, 2025-02-14:
It was a bit more complex to fix, but thanks for looking into it.
The fix will be part of the next release

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42