bernhard-42/vscode-ocp-cad-viewer #143: Change quickstart button to install build123d from pypi and not git

GarryBGoode, 2025-01-15:
Hi!

I noticed that the `quickstart with build123d` button of OCP VSCode installs the git version of build123d, which is by default the dev branch.

While currently it might be more stable than the pypi release due to a recent issue with cadquery-ocp, I think in the long term it would be better to use the pypi version of build123d for such a quickstart command.

At the moment the correct way to install build123d via pip is `pip install build123d "cadquery-ocp<7.8"`.

bernhard-42, 2025-01-16:
Will do as soon as build123d install is stable.
Feel free to change the commands yourself in the configuration of OCP VS Code

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42