bernhard-42/vscode-ocp-cad-viewer #145: Bug with showing rotated compounds

MatthiasJ1, 2025-01-27:
```py
thread = Rot(90)*TrapezoidalThread(20, 1, 38*2, 3)
circle = Rot(90)*Circle(10)
```
```py
show(thread)
```

[image]

```py
show(circle)
```

[image]

```py
show(thread, circle)
```

[image]

MatthiasJ1, 2025-01-27:
Could be the cause of https://github.com/gumyr/build123d/issues/811

bernhard-42, 2025-01-27:
Note to myself: In the array, thread is not treated as assembly, hence top level location gets lost

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42