bernhard-42/vscode-ocp-cad-viewer #146: View web server on other device broken

MatthiasJ1, 2025-01-31:
The web server works if running on the same machine. If run with `--host 0.0.0.0`, it can be accessed from another device. If I open `<computer ip>:3939/viewer` on my iPad, I will get the OCP viewer UI with the default OCP eye logo which I can view in 3D, but if I run `show()` on my computer, I get the following output on the OCP viewer server:
```
127.0.0.1 - - [31/Jan/2025 10:53:30] "GET / HTTP/1.1" 200 -
127.0.0.1 - - [31/Jan/2025 10:53:30] "GET / HTTP/1.1" 200 -
127.0.0.1 - - [31/Jan/2025 10:53:30] "GET / HTTP/1.1" 200 -
127.0.0.1 - - [31/Jan/2025 10:53:30] "GET / HTTP/1.1" 200 -
[2025-01-31 10:53:30,928] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/opt/homebrew/Caskroom/micromamba/envs/cad/lib/python3.13/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
  File "/opt/homebrew/Caskroom/micromamba/envs/cad/lib/python3.13/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/opt/homebrew/Caskroom/micromamba/envs/cad/lib/python3.13/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/homebrew/Caskroom/micromamba/envs/cad/lib/python3.13/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/opt/homebrew/Caskroom/micromamba/envs/cad/lib/python3.13/site-packages/flask_sock/__init__.py", line 62, in websocket_route
    f(ws, *args, **kwargs)
    ~^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Caskroom/micromamba/envs/cad/lib/python3.13/site-packages/ocp_vscode/standalone.py", line 220, in handle_message
    self.javascript_client.send(data)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'send'
```

bernhard-42, 2025-02-03:
I just tried it on my machines:

Linux:
```bash
pip install ocp_vscode
cd build123d
pip install .  # to avoid the HashCode problem
python -m ocp-vscode --host="0.0.0.0"
```
Same machine
```python
In [1]: from build123d import *
In [2]: from ocp_vscode import *
In [3]: b = Box(1,1,1)
In [4]: b = chamfer(b.edges(), 0.1)
In [5]: show(b)
UserWarning: The viewer doesn't seem to run
Port 3939 is open, using it
Jupyter console not installed
+
```

Other machine
```python
In [1]: from build123d import *
In [2]: from ocp_vscode import *
In [3]: set_port(3939, host='192.168.178.41')
In [4]: b = Box(1,2,3)
In [5]: b = chamfer(b.edges(), 0.2)
In [6]: show(b)
+
```

In both cases the box will be shown in the viewer as expected

bernhard-42, 2025-02-03:
And yes, the `UserWarning: The viewer doesn't seem to run` should not appear

MatthiasJ1, 2025-02-03:
I think your case might be slightly different. You need to open the viewer in the browser on `Other machine` while the server is running on `Linux`

bernhard-42, 2025-02-03:
I did:
- `python -m ocp_vscode --host="0.0.0.0"` on the remote Linux server (headless, no monitor)
- On my Mac I run the browser and connect to the viewer on the Linux machine
- Both from the Mac and the Linux server I can show objects.
- I then opened the viewer from my iPad and again, when showing objects from both machines, they update the viewer on the iPad

Am I missing someting?

bernhard-42, 2025-02-07:
@MatthiasJ1 I now show
```bash
No browser registered. Please open the viewer in a browser or refresh the viewer page
``` 
in the terminal of the viewer server when this error occurs.
Hope that helps

bernhard-42, 2025-03-02:
Closing for now, since I have no idea what else to do

closed 2025-03-02 by bernhard-42

MatthiasJ1, 2025-03-04:
I apologize for the silence, I was waiting to try this on a different network to rule that out and then it slipped through. Here are explicit reproduction steps. Both scenarios are done in a clean environment with other instances of the viewer running.

## Scenario 1: Works as expected

`mac_1 > TTY_1`
```
python -m ocp_vscode --host="0.0.0.0"
```

`mac_1 > TTY_2`
```
open http://$MAC_1_IP:3939/viewer
# mac_1 > TTY_1 outputs "GET ok", "Browser as viewer client registered", and browser shows Eye Logo

python -c 'import build123d, ocp_vscode; ocp_vscode.show(build123d.Sphere(1))'
# Browser shows sphere
```

## Scenario 2: Does not work

`mac_1 > TTY_1`
```
python -m ocp_vscode --host="0.0.0.0"
```

`mac_2 > TTY_1`
```
open http://$MAC_1_IP:3939/viewer
# mac_1 > TTY_1 outputs "GET ok", and browser shows Eye Logo
# NOTE:  "Browser as viewer client registered" is missing
```

`mac_2 > TTY_2`
```
python -c 'import build123d, ocp_vscode; ocp_vscode.show(build123d.Sphere(1))'
# Browser does nothing
# mac_1 > TTY_2 outputs "No browser registered. Please open the viewer in a browser or refresh the viewer page"
```

Theoretically, this should work exactly the same as Scenario 1. It sounds like this might be working on your network. If I had to guess I would say some part of the client registration code makes assumptions about the network that are true for your network but are not general.

My original guess was that the client expects the server to be on localhost. When you said that it works for your remote server, that ruled out that issue. However, I just checked the console output for the viewer on `mac_2` and it says web socket failed to connect to localhost. Are you port forwarding your remote server to localhost?

bernhard-42, 2025-03-05:
I do not have any port forwarding, etc, just a Linux server in the LAN (banach) and a Mac as my working desktop (atiyah).
No idea why in your case the browser doesn't register. Here a demonstration on my machines:

### Starting scenario:

[image]

As you can see, on the top the Linux box has a different ip than the Mac box at the bottom.

### Then, clicking on `http://192.168.178.41:3939/viewer`

The browser registers and shows the logo

[image]

### Then, showing a box from the Mac

[image]

### Then, stopping the server and restarting it

[image]

Trying to show a Cylinder fails

[image]

### Then, refreshing the browser

[image]

Client registered and shows the logo

### Finally, showing the cylinder

[image]

bernhard-42, 2025-03-05:
When I press refresh in the browser again with dev tools open, one can see the websocket call to the Linux server.
`L:{}` is the registration message:

[image]

[image]

[image]

MatthiasJ1, 2025-03-11:
I've found the issue.

In your case, notice that when you specify 0.0.0.0, it the viewer claims to be running at 192.168.178.41:
```
python -m ocp_vscode --host 0.0.0.0
The viewer is running at http://192.168.178.41:3939/viewer
```

Here is what I get:
```
python -m ocp_vscode --host="0.0.0.0"   
The viewer is running at http://127.0.0.1:3939/viewer
```

The client is trying to connect to whatever url the host claims to be serving. 0.0.0.0 means serve all of my IPs, which means the server is running both at 127.0.0.1 and 192.168.178.41 simultaneously, however, the client, server, and `ocp_vscode.send` do not understand this. You will see this fact if you look at the rest of your Flask output:

<pre>
> python -m ocp_vscode --host 0.0.0.0  
The viewer is running at http://192.168.178.41:3939/viewer

 * Serving Flask app 'ocp_vscode.standalone'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * <b>Running on all addresses (0.0.0.0)</b>
 * <b>Running on http://127.0.0.1:3939</b>
 * <b>Running on http://192.168.178.41:3939</b>
</pre>

So if I specify `python -m ocp_vscode --host 192.168.178.41`, it will now work from another machine, but `localhost:3939/viewer` will break.

Additionally, if you specify the host ip manually, the default localhost `ocp_vscode.set_port` will also stop working and it needs to be pointed to the external IP which should not be necessary.

bernhard-42, 2025-03-12:
I have not a lot of knowledge around networks and cannot reproduce.

This is what I use:

```python
if host == "0.0.0.0":
            hostname = socket.gethostname()
            host = socket.gethostbyname(hostname)
```

Doesn't seem to work on your machine. 
I guess I need to send `0.0.0.0` directly to flask, but what IP to add then to the output in order to allow clicking on it in the terminal to open (eg iterm allows this)
Need to research.

Could you please check on your machine that providing flask with `0.0.0.0` will work?
As I said, I cannot reproduce

bernhard-42, 2025-03-12:
perplexity says that this is the right way:
```
import socket

def get_local_ip():
    with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as s:
        # Connect to a public IP (doesn't send data)
        s.connect(("8.8.8.8", 80))
        return s.getsockname()[0]

ip_address = get_local_ip()
print(f"Local IP Address: {ip_address}")
```

Would that work on your machine?
I guess it will fail when there is no internet access, though

MatthiasJ1, 2025-03-13:
That is not the issue:

1. The server is running under two IP addresses: loop, lan
2. The user provides a server IP to the client (i.e. URL in browser or target host in python)
3. The client connects to host (this is obvious in the case of the browser since you still see the viewer but it doesn't work)
4. The client asks the host what IP should I use for web socket.
5. The host returns the IP it has gotten from flask. This is randomly (but seemingly deterministic based on the system) either loop or lan.
6. The client tries to open the web socket IP given to it by the host which can be incorrect, e.g. my case where I am trying to connect to lan but the server returns loop. In your case, the server coincidentally returns lan which works for your mac but it will not work for your server.

You have already reproduced this bug: if you look your first response, under your "Same Machine" output it says
```
UserWarning: The viewer doesn't seem to run
```

The solution is to simply use the IP used to first connect to the server for the web socket connection rather than asking the server for an IP to use.

MatthiasJ1, 2025-03-13:
> Could you please check on your machine that providing flask with 0.0.0.0 will work?

As opposed to `"0.0.0.0"`? Yes, that will have the same result.

> As I said, I cannot reproduce

You have already reproduced this bug: if you look your first response, under your "Same Machine" output it says
```
UserWarning: The viewer doesn't seem to run
```

> perplexity says that this is the right way:

That code is contradicts itself.

Either way the solution is to use the connection IP rather than having the server give out one of its multiple IPs.

bernhard-42, 2025-03-13:
> In your case, the server coincidentally returns lan which works for your mac but it will not work for your server.

Well, it worked "coincidentally" for both the Mac and my Linux server. For Linux see my protocol above.

> This is randomly (but seemingly deterministic based on the system) either loop or lan.

I read about it, it is not deterministic under Linux, but depends on the Linux distribution. Has nothing to do with Mac vs. Linux. 
I test all features on Mac, Windows and Ubuntu 22.04. That is all I can do.

> As opposed to "0.0.0.0"? Yes, that will have the same result.

This is not what I mean: I currently give flask what I receive from the wrong way to determine the IP address `app.run(host=str(ip))` . I can instead give `app.run(host='0.0.0.0')` And I wanted to know whether flask then listens to both IP addresses

> That code is contradicts itself.

nope, it works. It might need internet access, so I will not use it. As I said I am not a network expert

_Btw., not sure it helps that your language starts to sound a bit ironic (but maybe I misinterpret it). This is a free tool I build and maintain in my free time._

bernhard-42, 2025-03-13:
> You have already reproduced this bug: if you look your first response, under your "Same Machine" output it says

No, this is a different issuse. Please read the context.

bernhard-42, 2025-03-13:
I reread everything again and yes, I misunderstood the situation:
flask indeed gets "0.0.0.0", so all good. The issue is the IP address for the websocket connection.
I was chasing a completely different problem, sorry

MatthiasJ1, 2025-03-13:
I don't mean to be ironic, just trying to be to the point, sorry if I came off that way. I appreciate your efforts on this tool and your willingness to look into the issues I open.

By windows vs mac I was just referring to your two computers, not the OS itself.

By the code contradicting itself I mean it connects to the internet to determine its lan IP (although I just looked into this and it doesn't complete the connection so it technically doesn't really matter what IP you use here so never mind).

bernhard-42, 2025-03-13:
No worries, all good.

When I change `index` in the class `Viewer` (where the web page is rendered) to:
```python
    def index(self):
        # The browser will connect with an ip/hostname that is reachable from remote.
        # Use this ip/hostname for the websocket connection
        address, port = request.host.split(":")

        return render_template(
            "viewer.html",
            standalone_scripts=SCRIPTS,
            standalone_imports=STATIC,
            standalone_comms=COMMS(address, port),
            standalone_init=INIT,
            styleSrc=CSS,
            scriptSrc=JS,
            treeWidth=self.config["tree_width"],
            **self.config,
```

I use as you suggest the network reachable address for the websocket communication.
Afterwards I can access the viewer from the server machine via localhost and from another machine via the server ip address or name.

Would that work for you?

A remaining issue is, how tp properly determine one or more complete viewer URLs like 
`http://192.168.178.41:3939/viewer`, `http://127.0.0.1:3939/viewer` to help the user

bernhard-42, 2025-03-13:
btw. one also needs to `from flask import Flask, render_template, request`

bernhard-42, 2025-03-13:
I could use [ifaddr](https://github.com/ifaddr/ifaddr/tree/master)
```
import ifaddr

adapters = ifaddr.get_adapters()
for adapter in adapters:
    for ip in adapter.ips:
        if ip.is_IPv4:
            print(f"http://{ip.ip}:{port}/viewer")
```
when `--host`is `0.0.0.0`

Which on my server is 
```
http://127.0.0.1:3939/viewer
http://192.168.178.41:3939/viewer
http://172.17.0.1:3939/viewer
http://172.18.0.1:3939/viewer
```

At least you could cmd-click (Mac) on the right link in iterm. Maybe this also works in windows and Linux, idk

bernhard-42, 2025-03-13:
hmm, but I would need to find out how to omit the docker and network bridge addresses.
Networking is such a complicated stuff ...

MatthiasJ1, 2025-03-13:
The Flask startup log already gives you this:
```
* Running on all addresses (0.0.0.0)
* Runnina on htto://127.0.0.1:3939
* Running on http://192.168.178.41:3939
```

Personally I think just specifying the localhost url is enough. If you want to show the same message as flask does but with the /viewer attached, the way they do it is they just output localhost url and then again with the ip from the `get_local_ip()` snippet you linked. If it fails due to no network, they just skip it.

https://github.com/pallets/werkzeug/blob/7868bef5d978093a8baa0784464ebe5d775ae92a/src/werkzeug/serving.py#L833

bernhard-42, 2025-03-14:
Great finding, thanks!
Now I can have the same message as flask provides, just extended by the suffix `/viewer`