bernhard-42/vscode-ocp-cad-viewer #149: Regression: show_all() fails for ShapeList

MatthiasJ1, 2025-03-12:
For each of these, call `show_all()` for an empty scene. (This is because for the ones where it fails, nothing will appear to happen if it is already shown from a previous working operation, the empty scene is not required to trigger bug).

```py
s = Rectangle(5,5)
```
Works

```py
s = [Rectangle(5,5)]
```
Works

```py
s = ShapeList([Rectangle(5,5)])
```
Fails with:
```
show_all: Unknown type: <class 'OCP.TopoDS.TopoDS_Compound'>
```

bernhard-42, 2025-03-12:
This is bad, since this is in ocp-tessellate, I should be able to provide a quick fix (and add it to the tests, thought I had it covered ...)

bernhard-42, 2025-03-12:
Looked into it, I don't think it is a regression. I never had `ShapeLists` in mind that have a `Compound` as element.
None of the typical methods `.edges()`, `.faces()`, ... return a `Compound` afaik. 
But yes, `ShapeList` elements can be of type `Shape` or `Vector` and a `Compound` is a `Shape`. 
Btw. `ShapeList([Vector(1, 2, 3)])` also fails.

bernhard-42, 2025-04-13:
fixed in 2.7.0

closed 2025-04-13 by bernhard-42