bernhard-42/vscode-ocp-cad-viewer #150: Please release the extension to OpenVSX

fxdave, 2025-03-14:
Microsoft legally forbids usage of their marketplace in any open-source builds from VSCode.

Every Linux distribution uses open-source builds (free from microsoft's proprietary codes). Also forks, app containers, etc.. have to use OpenVSX extension store.

fxdave, 2025-08-25:
Where does it require your address exactly? 

 1. Visit this: https://www.open-vsx.org/
 2. Click on "publish"
 3. It says you must log in.  Then if you click on "log in" it starts a github authentication.

tfroment, 2025-08-26:
The Eclipse Publisher Agreement requires a mailing address because, under Belgian law (Eclipse Foundation is a Belgian AISBL), parties to a contract must be uniquely identifiable. The Eclipse Foundation Intellectual Property Policy also requires accurate contributor records for due diligence and audits. Your address is never made public, it is only stored internally for legal compliance.

References:
- Publisher Agreement FAQ: https://www.eclipse.org/legal/epl-2.0/faq.php#publisher-agreement
- Eclipse Foundation IP Policy: https://www.eclipse.org/org/documents/EclipseIPPolicy20100601.pdf

Best

jdegenstein, 2025-08-26:
Does Eclipse accept the address on paper by mail and then store it solely on paper?