bernhard-42/vscode-ocp-cad-viewer #152: Docker standalone viewer

errge, 2025-03-18:
Hey there!

First of all, let me say that this is an amazing project and will definitely use it for all my build123d and cadquery learning that I plan to do in the coming weeks.

I'd like to do a github actions integration, where my endgoal is to have automatically built docker containers that can be used in the standalone mode. (I'm not a vscode user.)

The end-user experience would be something like this: `docker run -it --rm -p 3939:3939 ghcr.io/image/name` and then after that they can just load up their python interpreter locally and use `osp_vscode.show` functions (and of course open the browser at localhost:3939).

My question regarding this work: I'm happy to do all this, but what is your preference regarding future maintenance and integration level.

1. I put this under my own organization, e.g. the docker image name will be `ghcr.io/nilcons-contrib/vscode-ocp-cad-viewer`, and I'd politely ask that you reference the docker command in your README
2. I create the integration (`.github` directory for actions) and send you a pull request to activate in your master branch, then the image name will be `ghcr.io/bernhard-42/vscode-ocp-cad-viewer`.

Pros/cons of the first solution vs second solution:

  - first solution pro: you don't have to ever maintain the build in the future, it's all on me,
  - first solution con: users might be worried that it's some third party docker image name, not bernhard-42,
  - second solution pro: nicely integrated into the project and now standalone users can just use the official image name coming from your ghcr,
  - second solution con: if I disappear, maintenance is on you.

errge, 2025-03-18:
This feature is developed now, you can preview it here: https://github.com/nilcons-contrib/vscode-ocp-cad-viewer?tab=readme-ov-file#standalone-mode-with-docker

Feel free to comment or tell me if/how do you want to go forward. Cheers!

bernhard-42, 2025-03-18:
Nice job, but I would prefer the first solution. I already have too many open source repos to maintain and will also remove docker support from Jupyter-Cadquery (no technology reason, mainly because of maintenance and testing efforts).
Happy to reference it in my Readme and you you can also announce it in the cadquery/build123d discord channel, if you want.

errge, 2025-03-19:
Thank you for taking a look.

As agreed: https://github.com/bernhard-42/vscode-ocp-cad-viewer/pull/153

Docker build is online and ready to go.  This issue can be closed once README change is pulled.

bernhard-42, 2025-04-13:
done in 2.7.0

closed 2025-04-13 by bernhard-42