bernhard-42/vscode-ocp-cad-viewer #155: Viewer Crashing

MatthiasJ1, 2025-03-21:
Recently I've been seeing the viewer crash more often even though how I am using it has not changed. I get the following view:

[image]

It is unresponsive but Viewer Manager says it's "Running". Would it be possible to have the viewer auto-restart itself and show an error message in this case? Currently you have to close the window, go to the OCP CAD VIEWER tab, and restart it manually.

MatthiasJ1, 2025-03-21:
This has actually started happening very frequently (once every 15min) but I don't get any error logs to report. I will try reinstalling.

bernhard-42, 2025-03-22:
Does the viewer crash (i.e. the gpu based canvas that renders the objects in webgl) or only the treeview (which is a html/javascript component)

MatthiasJ1, 2025-03-22:
The whole things stops working

bernhard-42, 2025-03-23:
Since I did not release anything, could it be the underlying browser engine or graphics driver that received an upgrade? Does it happen in both VS Code and standalone? They should use different browser engines.

And if I understand you correctly, no messages in the javascript dev tools, right?

MatthiasJ1, 2025-03-23:
After reinstalling everything it seems to be working. Idk, I checked the default VSCode tabs: Problems, Output, Debug Console, Terminal, Ports

bernhard-42, 2025-03-23:
Typically I check the VS Code developer tools, i.e. the analogy to the browser console. Can be toggled in VS Code command menu "toggle developer tools" or so.
This shows everything, not only what VS code exports into its tabs

bernhard-42, 2025-03-23:
So, let's hope it is working gain ...

MatthiasJ1, 2025-03-24:
Thanks, I haven't used vscode other than for cad. If the issue resurfaces I'll check those and submit the error logs.