bernhard-42/vscode-ocp-cad-viewer #157: "folder,true" prevent to create snippet file

bongiozzo, 2025-03-22:
https://github.com/bernhard-42/vscode-ocp-cad-viewer/blob/HEAD/docs/snippets.md

When trying to install snippets on MacOS
Default value for installation folder contains ",true" string which prevents installation with default values

[PROJECT_PATH],true/.vscode

bernhard-42, 2025-03-23:
Note to myself:
Change 
```
                let dotVscode = await vscode.window.showInputBox({
                    prompt: "Location of the .vscode folder",
                    value: `${getCurrentFolder()}/.vscode`
                });
```
to
```
                let curFolder = getCurrentFolder();
                if (curFolder[1]) { 
                    dotVscode = await vscode.window.showInputBox({
                        prompt: "Location of the .vscode folder",
                        value: `${curFolder[0]}/.vscode`
                    });
                } else {
                    return
                }
```

bernhard-42, 2025-04-13:
fixed in 2.7.0

closed 2025-04-13 by bernhard-42