bernhard-42/vscode-ocp-cad-viewer #161: How to fix settings.json

Paul8043, 2025-04-23:
I am running VSCODE-version 1.99.3. During startup, opening a model-folder, I get some complaints which say my installation-commands would not be up-to-date. It should be fixed in settings.json.

Directly after opening the model-folder I put the mouse-cursor inside Demo.py, then I get the error-messages captured here 

[image]

I want only to to use the newest OCP CAD VIEWER which is currently 2.7.1. But it seems that a further try is made to activate an older version named 2.4.1. This conflict can be seen in the next screen-dump:

[image]

It is totally unclear why the OCP CAD VIEWER with version 2.4.1 is listed in the LIBRARY-MANAGER-Section. Thought the OCP CAD VIEWER with version 2.7.1 in the VIEWER-MANGER-Section should be sufficient resp. this version should be also reflected in the LIBRARY-MANGER-Section. In my configuration there is always a mismatch between these two versions. I do not have any idea how this can be fixed in settings.json. I have found a settings.json, but this file does not contain any explict versions.

How to get rid of this this problem?

If I run the model (Demo.py) for the second time, then the model is correctly rendered in the OCP-CAD-VIEWER. But the error-message in lower right corner remains present.

[image]

bernhard-42, 2025-04-23:
There is the viewer version (the extension) and the python lib version (for the show command).
After updating the extension, VS Code checks the python version and wants to upgrade it. Unfortunately, I had to change the way the installation commands work. You seem to have a customized one in your .vscode/settings.json file in your project folder. Load it into the editor and search for the installation command block. Delete it and save it.
Now the python installation (update) in VS Code should work.

As a workaround, use `pip install --upgrade ocp_vscode==2.7.1`

... or so ... I am currently on my iPad and can't check the code

bernhard-42, 2025-04-23:
btw, the red marked,dialog in VS Code shows you the string to search for in settings.json

Paul8043, 2025-04-24:
Thanks for your first hints. I do not have a .vscode/settings.json in my project-folder. Could it be that settings.json resides in further folders? But I have no idea where to look for potential candidates. 

Maybe we go step back and check first whether my whole setup is still adequate. I work on Windows11 and I have created a special environment with mambaforge for my CAD-projects. The workflow was:

1. open a mamba-command-window
2. activate the environment for Build123d
3. cd my-project-directory
4. code    # start VS-Code from within the mamba-comand-window

This worked fine in the past. But now this is no longer true. If I list packages from the this environment, I find a

- python 3.11.4
- ocp-vscode 2.4.1
- build123d 0.6.1.dev66+g235260a

Here the packages ocp-vscode and build123d are out-of-date and probably the source of the problem. These are the packages, among some others, in my environment. The OCP CAD VIEWER does similiar things. And I hope the OCP CAD VIEWER should be the master of all of this. As I understood OCP CAD VIEWER takes also care for right python version.

What is the recommended sequence for the installation?
My perception is (could be wrong):

1. create a mamba-environment
2. code   # start VS-Code from within the mamba-command-window
3. inside VS-CODE try to install extentsion "OCP CAD Viewer"
4. inside the VIEWER MANGER do a "Quickstart build123d"

Let me know whether this correct or not? Let me also know what you recommend as best practice?

What is still unclear to me: If the OCP CAD Viewer loads some extensions, are they stored in the current environment or somewhere else? If they are stored in the environment, I would expect the see ocp-vscode upgraded to version 2.7.1. If they stored outside the environment, I need further explanation. Some months ago VS-CODE has explictly mentioned that the python-program is executed from within the environment. Now this indication is gone.

bernhard-42, 2025-04-24:
Click on the gear in the ocp-vscode side bar

[image]

You'll see this

[image]

Add the string " install" behind "OCP CAD Viewer"

[image]

Click on "Edit settings.json" under "Ocp Cad Viewer  > Advanced: Install Commands"

This is how it should look like:

[image]

Delete this complete block from `"OcpCadViewer.advanced.installCommands: {"` to the final `}`
Delete the comma in the line above, if it is the last entry.
Save and restart the editor.

When you after restart go to VS Code's Library Manager, you should be able to upgrade ocp_vscode.

btw. you're description looks ok to me.

> What is still unclear to me: If the OCP CAD Viewer loads some extensions, are they stored in the current environment or somewhere else?

VS Code extensions are a complete different topic (typescript packages) and stored independent of any Python stuff.
That is the reason, why after upgrading VS Code extension from VS Code marketplace, the extension needs to upgrade the Python packages ocp_vscode of the same version. 

So `show` from ocp_vscode runs in the Python environment. It uses a websocket connection to the VS Code extension, i.e. the two worlds Python and VS Code extension are completely separate, they just communicate with each other by using websockets.

Does that help?

Paul8043, 2025-04-25:
Thanks for guiding me to the right place. Up to "OCP CAD Viewer install" everything behaves as you have described. After that step my screen look like this:

[image]

It seems that the settings have been changed for User and for Workspace, and therefore they should be fixed twice. Just for completeness I give the current settings.json for User and Workspace:

[UserSettingsJson.txt](https://github.com/user-attachments/files/19904132/UserSettingsJson.txt)

[WorkspaceSettingsJson.txt](https://github.com/user-attachments/files/19904151/WorkspaceSettingsJson.txt)

Now I am trying to remove the block "OcpCadViewer.advanced.installCommands". For the Workspace this leaves an almost empty file with a single pair of curly braces. For the User the contents deviates from your expectation, it contains much more and probably very old information. Unclear to me why this has survived.

I got an editor-window inside VS-CODE where I could change the json-code. However Ctl-S does not save the new contents. Save from the File-Menu does not help either. How to make these changes persistent? Or should I change these two files outside VS-CODE with a normal text-editor?

After trying these changes I have got a ".vscode/settings.json" in my project-folder. Was not present yesterday!

bernhard-42, 2025-04-25:
settings.json can indeed be empty ("{}") which just means that there are no workspace or user specific settings  (and yes I have forgotten that there are two locations).

OCP CAD Viewer uses the official VS Code settings APIs like others. This means that all simple settings can be changed in the settings UI and all complex (hierarchical objects) in the settings.json files.
As such, there are probably many more entries in the settings.json files, however, many are not "owned" by my viewer. 
Everything owned by my viewer starts with "OcpCadViewer".
Only the install commands block needs to be deleted.

I would have expected that you can save changed setting files from VS Code. In fact, this is how I do it. No idea why it doesn't work on your machine, maybe VS Code needs a restart?

And sorry for the hassle, I had to implement a lot of workarounds in the past for missing features. Now that all parts (build123d, cadquery, OCP, and of course my viewer ...) get more mature, I try to clean up, but not every clean up is as easy for the user as I had hoped ...

Paul8043, 2025-04-26:
The problem is more subtle. I have checked the editing of the settings more carefully. VS-Code changes the settings instantly, a special save-action like CTRL-S is not needed! I have checked the contents of the settings-files. The ocp-install-command-section has gone and I could reach the state you have proposed.  However after closing the project-folder and reopening it gain the old settings reappear magically. The settings-files are changed under the hood! Is there a syncronization-mechanism for settings? I do not have anything activated in this area, and I am even not aware what might be considered as a master in this case. Does anybody know how to disable this feature? This explains at least why I cannot fix the version-conflict. Maybe it is better to continue with a fresh installation of VS-CODE. Or are there any better ideas?

bernhard-42, 2025-04-26:
Great that it works now, and sorry for the hassle. 
I also don't understand everything VS Code does ...
But glad you could resolve it.

bernhard-42, 2025-06-26:
Closing for now. If you still have an issue, please reopen

closed 2025-06-26 by bernhard-42