bernhard-42/vscode-ocp-cad-viewer #163: Ignore "No editor open"

MatthiasJ1, 2025-05-06:
About 30% of the time when I open VSCode the `OCP CAD Viewer` window will be blank and I will get the error `No editor open`. I then have to either restart VSCode or close the `OCP CAD Viewer` window and then press `OCP CAD Viewer: Open viewer (button)` in the `View Manager` tab.

Could we just assume if the `OCP CAD Viewer` window is already open upon launch, it should just startup regardless of whether it detects a Build123d editor rather than throwing an error and displaying an empty tab?

bernhard-42, 2025-05-06:
yes, I know, and I am already working on it.
The current behavior was meant to help with auto-detection of Python env and viewer port, but it turned out to be more annoying than helpful.
I hope to come up with something much easier and more user friendly soon ...

MatthiasJ1, 2025-05-06:
Sounds good, thanks!

MatthiasJ1, 2025-05-11:
One related note, if you switch to a tab that uses B3D it will auto-open the vscode viewer. If you are using an external viewer in the browser, this keeps getting in the way.

bernhard-42, 2025-05-11:
You can turn auto open off in the VS Code settings

MatthiasJ1, 2025-05-11:
Didn't know that, thanks!

Paul8043, 2025-05-14:
I would also appreciate very much, if  the "no editor open" could go away completely. That would be the icing on the cake!

bernhard-42, 2025-05-14:
it is coming - just testing all scenarios and all platforms.

bernhard-42, 2025-06-26:
Fixed in 2.8.5

MatthiasJ1, 2025-06-28:
This has actually gotten worse for me in the latest release. My cad `.py` files actually don't have `include build123d` in them if that makes a difference.