bernhard-42/vscode-ocp-cad-viewer #166: add support for uv based environmnts

RonnyPfannschmidt, 2025-05-09:
i tried this on a virtualenv created by uv without pip preinstalled - everything broke until i added pip to the env

bernhard-42, 2025-05-09:
yes currently pip is required, see Readme

bernhard-42, 2025-05-09:
uv pip is incompatible in one feature I need, see https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/127

RonnyPfannschmidt, 2025-05-09:
Would you accept a importlib metadata based implementation that works for all python versions?

bernhard-42, 2025-05-09:
What I currently show that `uv pip` does not provide is the installer info. According to my research, importlib metadata does not reliably show this info (but I might be wrong). 
I don't want to maintain some Python hack just to support uv. Maybe I set installer simply to `?` when I don't receive it.

Since more people start to ask, I'll put it on the roadmap to somehow enable it.

But until then `uv pip install pip` is your friend

RonnyPfannschmidt, 2025-05-10:
Actually i wonder if it ain't be possible to ask the vscode python support for these details

bernhard-42, 2025-05-11:
I tried it, but it gets really complicated and I don't get "uv pip list" reliably to detect the right env from within the extension.
I will add `pip` to the list of dependencies, so that uv based environments work ootb.
Sorry, not willing to invest more in the Python packaging mess.