bernhard-42/vscode-ocp-cad-viewer #170: Spline fails to show

MatthiasJ1, 2025-06-14:
```py
from build123d import *
from ocp_vscode import *

poles=[Vector(*p) for p in [[0, -116], [88, -117], [85, -47], [87, 0]]]
s = Bezier(*poles).edges()
s += mirror(s)
s += mirror(s, Plane.YZ)

show_all()
```
Gives Error:
```
show_all: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.BRepAdaptor.BRepAdaptor_Curve()
    2. OCP.BRepAdaptor.BRepAdaptor_Curve(E: OCP.TopoDS.TopoDS_Edge)
    3. OCP.BRepAdaptor.BRepAdaptor_Curve(E: OCP.TopoDS.TopoDS_Edge, F: OCP.TopoDS.TopoDS_Face)
```

Spline is valid. Adding the following line prior to calling show_all() will fix it.
```py
s = Sketch() + s
```

bernhard-42, 2025-06-14:
Interesting. Glad there is a workaround.
Will try to squeeze the fix into 2.8.2 which is planned in 1-2 weeks (currently on vacation)

MatthiasJ1, 2025-06-15:
Thanks, enjoy your vacation!

bernhard-42, 2025-06-22:
Interesting, the first mirror leads to a `Curve` (inherited from `Compound`), i.e. a `Compound` with an `Edge`.
The second mirror also leads to a `Curve`, but this time it is a `Compound` of `Compounds`, and ocp-tessellate fails on this.

closed 2025-06-26 by bernhard-42