bernhard-42/vscode-ocp-cad-viewer #171: Random blank window in vscode 1.101.0

javimixet, 2025-06-15:
Testing ocp_vscode 2.8.1. After updating vscode from 1.100.3 to 1.101.0, sometimes, ocp_vscode starts to a blank window. 
Standalone works as expected. 
Next are vscode logs from blank window and successful starting respectively (ellipsis are long lists of "Port  * already in use")  :

```
[11:50:31 AM.710} DEBUG ] OCP CAD Viewer dispose
[11:50:31 AM.710} DEBUG ] OCPCADController dispose
[11:50:31 AM.713} INFO  ] Client disconnected
[11:50:31 AM.713} DEBUG ] Listener deregistered
[11:50:31 AM.717} DEBUG ] ~/.ocpconfig = {"version":2,"services":{}}
[11:50:31 AM.918} INFO  ] Server is shut down
[11:50:36 AM.336} DEBUG ] Set viewerStarting
[11:50:36 AM.336} DEBUG ] Start ocpCadViewer.ocpCadViewer for undefined
[11:50:36 AM.339} INFO  ] PythonPath: '/home/i/New_SD/build123d_venv/bin/python', environment: VirtualEnvironment, build123d_venv
[11:50:36 AM.750} INFO  ] ocp_vscode library version 2.8.1 matches extension version 2.8.1
[11:50:36 AM.750} INFO  ] Port 3939 already in use
[11:50:36 AM.751} INFO  ] Port 3940 already in use

[...]

[11:50:41 AM.654} INFO  ] Port 49150 already in use
[11:50:41 AM.654} INFO  ] Port 49151 already in use
[11:50:41 AM.655} INFO  ] Server started on port 49152
[11:50:41 AM.655} INFO  ] Starting websocket server ...
[11:50:41 AM.655} DEBUG ] Creating new webview panel
[11:50:41 AM.656} DEBUG ] Updating webview
[11:50:41 AM.658} DEBUG ] OCP Cad Viewer port: 49152, folder: /home/i/New_SD/build123d_venv/vscode_workspace, /home/i/New_SD/build123d_venv/vscode_workspace
[11:50:41 AM.659} DEBUG ] Command OCP CAD Viewer registered
[11:50:41 AM.661} DEBUG ] Unset viewerStarting
[11:50:42 AM.014} DEBUG ] Text editor changed: undefined
[11:50:42 AM.020} DEBUG ] Text editor changed: /home/i/New_SD/build123d_venv/vscode_workspace/ocp_vscode_demo.py
[11:50:42 AM.030} DEBUG ] ~/.ocpconfig = {"version":2,"services":{"49152":""}}
[11:50:44 AM.502} INFO  ] Client connected
[11:50:44 AM.503} DEBUG ] Listener registered
[11:50:59 AM.526} INFO  ] Client connected
[11:50:59 AM.556} INFO  ] Client disconnected
[11:50:59 AM.580} INFO  ] Client connected
[11:50:59 AM.626} INFO  ] Client disconnected
[11:50:59 AM.633} INFO  ] Client connected
[11:50:59 AM.659} INFO  ] Client disconnected
[11:50:59 AM.673} INFO  ] Client connected
[11:50:59 AM.709} INFO  ] Client disconnected
```

```
[11:44:42 AM.243} DEBUG ] OCP CAD Viewer dispose
[11:44:42 AM.243} DEBUG ] OCPCADController dispose
[11:44:42 AM.246} INFO  ] Client disconnected
[11:44:42 AM.246} DEBUG ] Listener deregistered
[11:44:42 AM.249} DEBUG ] ~/.ocpconfig = {"version":2,"services":{}}
[11:44:42 AM.450} INFO  ] Server is shut down
[11:44:47 AM.864} DEBUG ] Text editor changed: /home/i/New_SD/build123d_venv/vscode_workspace/ocp_vscode_demo.py
[11:44:47 AM.864} DEBUG ] Conditionally open viewer for /home/i/New_SD/build123d_venv/vscode_workspace/ocp_vscode_demo.py
[11:45:14 AM.137} DEBUG ] Set viewerStarting
[11:45:14 AM.137} DEBUG ] Start ocpCadViewer.ocpCadViewer for undefined
[11:45:14 AM.139} INFO  ] PythonPath: '/home/i/New_SD/build123d_venv/bin/python', environment: VirtualEnvironment, build123d_venv
[11:45:14 AM.509} INFO  ] ocp_vscode library version 2.8.1 matches extension version 2.8.1
[11:45:14 AM.509} INFO  ] Port 3939 already in use
[11:45:14 AM.510} INFO  ] Port 3940 already in use
[11:45:14 AM.511} INFO  ] Port 3941 already in use

[...]

[11:45:29 AM.037} INFO  ] Port 49149 already in use
[11:45:29 AM.037} INFO  ] Port 49150 already in use
[11:45:29 AM.037} INFO  ] Port 49151 already in use
[11:45:29 AM.037} DEBUG ] OCP Cad Viewer port: 49152, folder: /home/i/New_SD/build123d_venv/vscode_workspace, /home/i/New_SD/build123d_venv/vscode_workspace
[11:45:29 AM.037} DEBUG ] Command OCP CAD Viewer registered
[11:45:29 AM.037} DEBUG ] Unset viewerStarting
[11:45:29 AM.365} DEBUG ] Text editor changed: undefined
[11:45:29 AM.370} DEBUG ] Text editor changed: /home/i/New_SD/build123d_venv/vscode_workspace/ocp_vscode_demo.py
[11:45:29 AM.557} DEBUG ] ~/.ocpconfig = {"version":2,"services":{"49152":""}}
[11:45:30 AM.564} DEBUG ] ~/.ocpconfig = {"version":2,"services":{"49152":""}}
[11:45:31 AM.243} INFO  ] Client connected
[11:45:31 AM.244} DEBUG ] Listener registered
[11:46:01 AM.118} INFO  ] Client connected
[11:46:01 AM.121} INFO  ] Client disconnected
[11:46:01 AM.122} INFO  ] Client connected
[11:46:01 AM.124} INFO  ] Client disconnected
[11:46:01 AM.125} INFO  ] Client connected
[11:46:01 AM.126} INFO  ] Client disconnected
[11:46:01 AM.127} INFO  ] Client connected
[11:46:01 AM.128} INFO  ] Client disconnected
[11:46:01 AM.128} INFO  ] Client connected
[11:46:01 AM.129} INFO  ] Client disconnected
[11:46:01 AM.130} INFO  ] Client connected
[11:46:01 AM.130} INFO  ] Client disconnected
[11:46:01 AM.131} INFO  ] Client connected
[11:46:01 AM.132} INFO  ] Client disconnected
[11:46:01 AM.158} INFO  ] Client connected
[11:46:01 AM.160} DEBUG ] Received a new model
[11:46:01 AM.160} DEBUG ] {}
[11:46:01 AM.161} DEBUG ] Posted model to view
[11:46:01 AM.161} INFO  ] Client disconnected
[11:46:01 AM.163} DEBUG ] ~/.ocpconfig = {"version":2,"services":{"49152":""}}
[11:46:01 AM.163} INFO  ] Client connected
[11:46:01 AM.164} DEBUG ] Model data sent to the backend
[11:46:01 AM.164} INFO  ] Client disconnected
```

bernhard-42, 2025-06-15:
it doesn't find a free port, hence the long list in the middle. Which OS do you have?

javimixet, 2025-06-15:
Manjaro. Standalone finds it right away, though:
```sh
vscode_workspace ) "/home/i/New_SD/build123d_venv/bin/python3" -m ocp_vscode --theme
 dark

The viewer is running on http://127.0.0.1:3939/viewer

 * Serving Flask app 'ocp_vscode.standalone'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:3939
Press CTRL+C to quit
```

```sh
[I] / ) inxi -G
Graphics:
  Device-1: NVIDIA TU106M [GeForce RTX 2060 Mobile] driver: nvidia v: 570.144
  Device-2: Advanced Micro Devices [AMD/ATI] Renoir [Radeon Vega Series /
    Radeon Mobile Series] driver: amdgpu v: kernel
  Device-3: Luxvisions Innotech HP Wide Vision HD Camera driver: uvcvideo
    type: USB
  Display: x11 server: X.Org v: 21.1.16 driver: X: loaded: amdgpu,nvidia
    unloaded: modesetting,nouveau dri: radeonsi gpu: amdgpu
    resolution: 1920x1080~60Hz
  API: EGL v: 1.5 drivers: nouveau,nvidia,radeonsi,swrast
    platforms: gbm,x11,surfaceless,device
  API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: amd mesa v: 25.0.5-arch1.1
    renderer: AMD Radeon Graphics (radeonsi renoir ACO DRM 3.61
    6.12.28-1-MANJARO)
  API: Vulkan v: 1.4.313 drivers: nvidia,radv surfaces: N/A
  Info: Tools: api: eglinfo, glxinfo, vulkaninfo gpu: nvidia-smi
    x11: xdpyinfo, xprop, xrandr
```

bernhard-42, 2025-06-21:
One of the changes I introduced was to check ipv6 additional to ipv4 for finding used ports.
Does your machine support ipv6 or is it disabled (just in case you know)

javimixet, 2025-06-21:
In my machine, ipv6 is enabled:
```sh
[I] i ) cat /proc/sys/net/ipv6/conf/all/disable_ipv6  
0
```
Just downloaded vscode 1.100.3 and the viewer works. It complains about the jupyter extension, in case it's related:

[image]

1.101.1 works on 1 in ~20 tries.

Edit: when trying to reinstall jupyter extension on 1.100.3, it asks me to update vscode and don't let me reinstall the extension, and now the same behavior (long list of ports in use and blank ocp_vscode window) in this version too.

[image]

javimixet, 2025-06-22:
The start was reliable, even with 2.8.1, before 1.101.1 update. Start ocp_vscode automatically at vscode start almost never worked, but closing its window and relaunching worked every time.

Uninstalled 2.8.1 and installed 2.7.1. The problem with finding ports seems gone, but the blank window one remains:

[image]

bernhard-42, 2025-06-23:
@javimixet I have published (still as [preview](https://github.com/bernhard-42/vscode-ocp-cad-viewer/releases/tag/v2.8.2)) 2.8.2 which added some changes to make startup more reliable and significantly improved logging.
Maybe you could give it a try and see whether it works? And if not paste the output log again?
Thanks

javimixet, 2025-06-23:
Ok. I'll download and test it right now.
Here's the output of developer tools for 2.8.1:

[image]

bernhard-42, 2025-06-23:
Thanks, again, nothing unusual.
What a strange bug ...

javimixet, 2025-06-23:
2.8.2 asks to install build123d every time vscode is restarted or window reloaded.
If I leave the default port (3939) in the settings, it takes a long time to start. Changing it to 49152 avoids the long start.
Still leading to a blank window. If show_all() is called, throws this output in console (similar to 2.8.1):
```python
show_all: Type <class 'types.UnionType'> for name RotationLike cannot be visualized
show_all: Type <class 'types.UnionType'> for name VectorLike cannot be visualized
show_all: Cannot access viewer config. Is the viewer running?
('Cannot access viewer status. Is the viewer running?\n("\'str\' object has no attribute \'get\'",)',)
```
Here's the ocp_vscode log:
```ini
2025-06-23 19:01:37.983 [debug] extension.Activate: OCP CAD Viewer
2025-06-23 19:01:37.983 [info ] Successfully registered CadqueryViewer Status Manager
2025-06-23 19:01:37.984 [info ] Successfully registered CadqueryViewer Library Manager
2025-06-23 19:01:38.463 [debug] extension.Activate: Async start viewer /home/i/New_SD/build123d_venv/vscode_workspace/.venv/vscode_workspace/pinched_surf.py
2025-06-23 19:01:38.545 [debug] extension.conditionallyOpenViewer: Conditionally open viewer for /home/i/New_SD/build123d_venv/vscode_workspace/.venv/vscode_workspace/pinched_surf.py
2025-06-23 19:01:38.913 [debug] Text editor changed: /InteractiveInput-1
2025-06-23 19:01:39.090 [debug] Text editor changed: undefined
2025-06-23 19:01:39.099 [debug] Text editor changed: /InteractiveInput-1
2025-06-23 19:01:39.107 [debug] Text editor changed: undefined
2025-06-23 19:01:39.117 [debug] Text editor changed: /InteractiveInput-1
2025-06-23 19:01:39.489 [debug] Text editor changed: /home/i/New_SD/build123d_venv/vscode_workspace/.venv/vscode_workspace/pinched_surf.py
2025-06-23 19:01:39.489 [debug] extension.conditionallyOpenViewer: Viewer is already starting
2025-06-23 19:02:04.411 [debug] Text editor changed: extension-output-bernhard-42.ocp-cad-viewer-#1-OCP CAD Viewer Log
2025-06-23 19:02:11.462 [debug] Text editor changed: /home/i/New_SD/build123d_venv/vscode_workspace/.venv/vscode_workspace/pinched_surf.py
2025-06-23 19:02:11.462 [debug] extension.conditionallyOpenViewer: Viewer is already starting
2025-06-23 19:02:32.415 [debug] ocpCadViewer.ocpCadViewer: Set viewerStarting
2025-06-23 19:02:32.415 [debug] ocpCadViewer.ocpCadViewer: Start viewer for undefined
2025-06-23 19:02:32.417 [info ] PythonPath: '/home/i/New_SD/build123d_venv/bin/python', environment: VirtualEnvironment, build123d_venv
2025-06-23 19:02:33.277 [info ] extension.check_upgrade: ocp_vscode library version 2.8.2 matches extension version 2.8.2
2025-06-23 19:02:33.277 [debug] ocpCadViewer.ocpCadViewer: Starting OCPCADController
2025-06-23 19:02:33.278 [debug] OCPCADController.startCommandServer: Starting websocket server
2025-06-23 19:02:33.279 [info ] OCPCADController.startCommandServer: : Server listening on port 49152
2025-06-23 19:02:33.279 [debug] OCPCADController.start: Starting websocket server ...
2025-06-23 19:02:33.280 [debug] OCPCADViewer.createOrShow: Creating new webview panel
2025-06-23 19:02:33.283 [debug] OCPCADViewer.update: Updating webview
2025-06-23 19:02:33.284 [debug] OCPCADController.start: Starting backend ...
2025-06-23 19:02:33.284 [debug] Backend.startBackend: Use folder /tmp
2025-06-23 19:02:33.287 [debug] Backend.startBackend: Use python interpeter /home/i/New_SD/build123d_venv/bin/python
2025-06-23 19:02:33.287 [debug] Backend.startBackend: Terminal started
2025-06-23 19:02:33.288 [debug] ocpCadViewer.ocpCadViewer: OCPCADController started with port 49152 and folders: /home/i/New_SD/build123d_venv/vscode_workspace, /home/i/New_SD/build123d_venv/vscode_workspace/.venv/vscode_workspace
2025-06-23 19:02:33.306 [debug] ~/.ocpconfig = {"version":2,"services":{"49152":""}}
2025-06-23 19:02:33.421 [debug] Text editor changed: undefined
2025-06-23 19:02:33.746 [debug] Text editor changed: /home/i/New_SD/build123d_venv/vscode_workspace/.venv/vscode_workspace/pinched_surf.py
2025-06-23 19:02:33.949 [debug] ocpCadViewer.ocpCadViewer: Logo command succeeded
2025-06-23 19:02:33.949 [debug] ocpCadViewer.ocpCadViewer: Unset viewerStarting
2025-06-23 19:02:34.288 [debug] Backend.startBackend: Starting Python backend with delay 1000
2025-06-23 19:02:34.288 [info ] Backend.startBackend: Python backend running
2025-06-23 19:02:34.499 [debug] Viewer.log: resize listener registered
2025-06-23 19:02:34.555 [debug] Viewer.log: message listener registered
2025-06-23 19:02:38.992 [debug] OCPCADController.messages: message L received
2025-06-23 19:02:38.992 [debug] OCPCADController.messages: Python listener registered
2025-06-23 19:02:42.034 [debug] Text editor changed: extension-output-bernhard-42.ocp-cad-viewer-#1-OCP CAD Viewer Log
2025-06-23 19:02:53.661 [debug] Text editor changed: extension-output-bernhard-42.ocp-cad-viewer-#1-OCP CAD Viewer Log
2025-06-23 19:07:04.794 [debug] Text editor changed: undefined
2025-06-23 19:07:04.867 [debug] Text editor changed: /home/i/New_SD/build123d_venv/vscode_workspace/.venv/vscode_workspace/pinched_surf.py
2025-06-23 19:07:14.140 [debug] Text editor changed: undefined
2025-06-23 19:07:14.153 [debug] Text editor changed: /InteractiveInput-1
2025-06-23 19:07:17.983 [debug] OCPCADController.messages: message S received
2025-06-23 19:07:17.983 [debug] OCPCADController.messages: Received a config
2025-06-23 19:07:17.983 [debug] OCPCADController.messages: Posted config to view
[... cut ...]
2025-06-23 19:08:22.408 [debug] OCPCADController.messages: message C received
2025-06-23 19:08:22.411 [debug] OCPCADController.messages: message C received
2025-06-23 19:08:22.416 [debug] OCPCADController.messages: message C received
2025-06-23 19:08:26.421 [debug] Text editor changed: /InteractiveInput-1
2025-06-23 19:10:36.618 [debug] Text editor changed: undefined
2025-06-23 19:10:36.657 [debug] Text editor changed: extension-output-bernhard-42.ocp-cad-viewer-#1-OCP CAD Viewer Log
```
Outputs in developer tools look the same.

bernhard-42, 2025-06-23:
I can reproduce the port problem on Ubuntu. So I can fix this one.

bernhard-42, 2025-06-23:
I've solved the port issue (didn't handle Linux dual stack networking correctly).
I also have some blank windows, will continue to check.

bernhard-42, 2025-06-23:
And thanks for the clean OS test. Really appreciate your efforts and patience!

javimixet, 2025-06-23:
At my end the ports go smooth too with 2.8.3. Can't get rid of blank window, though. Disabled all extensions, created a new venv, a new profile, a new linux user, deleted .ocp_vscode (several times, because to downgrade from 2.8 to 2.7 seems necessary)... no luck. Also, standalone seems messed up since 2.8.2, it loads the model the first time it's called but not the next ones.

And thanks to you, man. This is a nice piece of software, and the last features are chef's kiss.

hmf, 2025-07-06:
I would just like to add that I have the same issue. I am just adding some additional info in case it is of use to anyone. Some differences  in regards to @javimixet 's report is (see below):

* Ubuntu
* Kernel 6.11
* OCP VSCode 2.8.5
* Using a virtual environment with the dev versions of build123d and CadQuery
 
I would also like to add the following:

* VSCode correctly starts the venv
* The OCP Cad Viewer correctly uses venv
* The OCP Cad Viewer correctly reports the venv libraries
* Starting the viewer standalone works correctly
* When using VSCode the ports are open
* Cleaning the VSCode cache did not seem to help 

However I get this error when running a script:

```shell
RuntimeError: Cannot access viewer config. Is the viewer running?
('Cannot access viewer status. Is the viewer running?\n("\'str\' object has no attribute \'get\'",)',)
```

And in fact that file does not exist:

```shell
~$ ls ~/.ocpconfig
ls: cannot access '/home/hmf/.ocpconfig': No such file or directory
```

Some questions:

Is this file created by the VSCode extension or the `ocp_vscode`? What is the expected content? Can I manually create it for testing/diagnosis? Can this missing file also result in an empty viewer?

For now I will use an external browser. However I believe making the viewer start more manual may facilitate diagnosis. For example starting the back-end and viewer in separate steps. For example, when Tensorboard starts it asks if the user wants and external or internal VSCode "browser".  

HTHs

Additional info:

```shell
(venv) hmf@gandalf:/tmp$ "/mnt/ssd2/hmf/cad/build123d/venv/bin/python" -m ocp_vscode --backend --port 3939
Viewer backend started
Logo model loaded
```

```shell
2025-07-06 08:32:28.701 [debug] extension.Activate: OCP CAD Viewer
2025-07-06 08:32:28.702 [info ] Successfully registered CadqueryViewer Status Manager
2025-07-06 08:32:28.704 [info ] Successfully registered CadqueryViewer Library Manager
2025-07-06 08:32:28.955 [debug] extension.Activate: Async start viewer undefined
2025-07-06 08:32:28.978 [error] extension.Activate: No editor focussed
2025-07-06 08:33:03.360 [debug] ocpCadViewer.ocpCadViewer: Setting viewerStarting
2025-07-06 08:33:03.360 [debug] ocpCadViewer.ocpCadViewer: Start viewer for undefined
2025-07-06 08:33:03.366 [info ] PythonPath: '/mnt/ssd2/hmf/cad/build123d/venv/bin/python', environment: VirtualEnvironment, venv
2025-07-06 08:33:04.104 [info ] extension.check_upgrade: ocp_vscode library version 2.8.5 matches extension version 2.8.5
2025-07-06 08:33:04.104 [error] ocpCadViewer.ocpCadViewer: No editor open
2025-07-06 08:33:04.104 [debug] ocpCadViewer.ocpCadViewer: Unsetting viewerStarting
2025-07-06 08:33:16.332 [debug] ocpCadViewer.ocpCadViewer: Setting viewerStarting
2025-07-06 08:33:16.332 [debug] ocpCadViewer.ocpCadViewer: Start viewer for undefined
2025-07-06 08:33:16.335 [info ] PythonPath: '/mnt/ssd2/hmf/cad/build123d/venv/bin/python', environment: VirtualEnvironment, venv
2025-07-06 08:33:17.087 [info ] extension.check_upgrade: ocp_vscode library version 2.8.5 matches extension version 2.8.5
2025-07-06 08:33:17.087 [error] ocpCadViewer.ocpCadViewer: No editor open
2025-07-06 08:33:17.087 [debug] ocpCadViewer.ocpCadViewer: Unsetting viewerStarting
2025-07-06 08:33:32.181 [debug] ocpCadViewer.ocpCadViewer: Setting viewerStarting
2025-07-06 08:33:32.181 [debug] ocpCadViewer.ocpCadViewer: Start viewer for undefined
2025-07-06 08:33:32.185 [info ] PythonPath: '/mnt/ssd2/hmf/cad/build123d/venv/bin/python', environment: VirtualEnvironment, venv
2025-07-06 08:33:32.903 [info ] extension.check_upgrade: ocp_vscode library version 2.8.5 matches extension version 2.8.5
2025-07-06 08:33:32.903 [error] ocpCadViewer.ocpCadViewer: No editor open
2025-07-06 08:33:32.903 [debug] ocpCadViewer.ocpCadViewer: Unsetting viewerStarting
[... 26 lines cut ...]
2025-07-06 08:33:46.361 [debug] Backend.startBackend: Starting Python backend with delay 1000
2025-07-06 08:33:46.361 [info ] Backend.startBackend: Python backend running
2025-07-06 08:33:47.611 [debug] ocpCadViewer.ocpCadViewer: Logo command succeeded
2025-07-06 08:33:47.611 [debug] ocpCadViewer.ocpCadViewer: Unsetting viewerStarting
2025-07-06 08:33:49.056 [debug] Viewer.log: resize listener registered
2025-07-06 08:33:49.109 [debug] Viewer.log: message listener registered
2025-07-06 08:33:55.279 [debug] OCPCADController.messages: message L received
2025-07-06 08:33:55.279 [debug] OCPCADController.messages: Python listener registered
2025-07-06 08:34:25.565 [debug] Text editor changed: extension-output-bernhard-42.ocp-cad-viewer-#1-OCP CAD Viewer Log
2025-07-06 08:36:27.700 [debug] Text editor changed: /mnt/ssd2/hmf/cad/build123d/example_66.py
```

```shell
~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
~$ uname -a
Linux name 6.11.0-29-generic #29~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jun 26 14:16:59 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
:~$ netstat | grep -i 3939
tcp        0      0 localhost:53836         localhost:3939          ESTABLISHED
tcp6       0      0 localhost:3939          localhost:53836         ESTABLISHED
```

hmf, 2025-07-06:
Just an additional note. I have noticed that if  I terminate all OCP viewer processes an run the modeling script I get:

```shell
  File "/mnt/ssd2/hmf/cad/build123d/venv/lib/python3.12/site-packages/ocp_vscode/config.py", line 516, in combined_config
    raise RuntimeError(
RuntimeError: Cannot access viewer config. Is the viewer running?
("Cannot access viewer status. Is the viewer running?\n('none',)",)
```

Note that this error is generated in the same line, but the output itself (last line) is different (see error above). This seems to suggest the app's state is different.

bernhard-42, 2025-07-06:
@hmf The `show` command and the viewer in VS Code and standalone communicate via websockets. So if no viewer runs, then `show` cannot get config info from the viewer and cannot show the object.

You can turn off autostart in the OCP CAD Viewer settings.

[image]

I am about to release a new version where the autostart on VS Code start will be removed. It was simply not possible to do this reliably. As long as autostart config is set, it will still start the viewer on saving or loading as long as the file contains a trigger statement. Please retest when the new version is published

javimixet, 2025-07-06:
@hmf 
6.11 will be EOL in four days, 6.14 should be available to install for Ubuntu 24.04, and maybe solves the blank window issue.
@bernhard-42 
Is the handling of log messages in standalone addressed in 2.8.5? I'm getting the same error output and supposed it wasn't, but  @hmf saying he is using an external browser made me doubt.

bernhard-42, 2025-07-10:
@javimixet @hmf 
I have uploaded 2.8.6 as preview to  https://github.com/bernhard-42/vscode-ocp-cad-viewer/releases/tag/v2.8.6

Maybe you try it to see whether it also works in your environment?
Thanks

bernhard-42, 2025-07-28:
@javimixet @hmf Does it still happen with 2.8.9 or can we close this issue?

closed 2025-07-29 by bernhard-42