bernhard-42/vscode-ocp-cad-viewer #177: "import build123d" required

MatthiasJ1, 2025-06-28:
The plugin currently refuses to work if
```py
import build123d
```
is not included in a the file (and the file is free of syntax errors).

I have a wrapper library that changes some behaviors of b3d so none of my cad files import it directly. If the file does not import build123d, it is no longer even possible to manually start the viewer as the menu to do so is hidden.

My opinion, and the reasoning behind #163, is that ocp_vscode should not care whether the user is actually using b3d, and it shouldn't automagically try to open/close itself depending on that.

[image]

bernhard-42, 2025-06-29:
> If the file does not import build123d, it is no longer even possible to manually start the viewer as the menu to do so is hidden.

That is simply a bug, the viewer shouldn't consider the content of the file to show its men.
The content should only be considered when autostart is on and a file contains a specific import.
I will change the triggering imports to be editable in the workspace settings for folks who like the feature and have a wrapper like you

> My opinion, and the reasoning behind https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/163, is that ocp_vscode should not care whether the user is actually using b3d, and it shouldn't automagically try to open/close itself depending on that.

I love that feature and use it all the time, so it'll stay. But as mentioned in https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/163, you can turn it off if it bothers you

MatthiasJ1, 2025-06-29:
> I will change the triggering imports to be editable in the workspace settings for folks who like the feature and have a wrapper like you

I appreciate that, thank you! I use VSCode exclusively for cad so I just leave it on permanently but that is a considerate feature to have anyway.

> I love that feature and use it all the time, so it'll stay.

Totally fair, I didn't realize it was optional.

> as mentioned in https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/163, you can turn it off if it bothers you

I forgot that you can do that (I even replied to you as such on that thread, smh). I just tested it out, however, and I am still running into the issue where sometimes when I open VSCode, the viewer won't start properly and I'll get an empty window (although now it seems to start more consistently than when auto-start is enabled):

[image]

bernhard-42, 2025-06-29:
The blank screen is a nasty bug. I really would like to get rid of it. 

More info always helps:

- What operating system do you use?
- How do you open it (with a folder, a file or none)
- How do you close VS Code (with the viewer on? Does the problem continue to exist when you close the viewer before closing VS Code?
- Or do you send your system to sleep and the blank screen appears when waking it up?

MatthiasJ1, 2025-06-29:
> What operating system do you use?

I use macOS 15.5

> How do you open it (with a folder, a file or none)

I have a projects folder where I keep all of my cad `.py` files and I open that as a VSCode workspace.

> How do you close VS Code

I just quit the app, leaving everything as-is. I have never had it fail by pressing the `launch viewer` button but I very rarely use that button since its faster to just quit and re-open VSCode than close the tab then navigate to that menu.

> Or do you send your system to sleep and the blank screen appears when waking it up?

I have never had this happen.

bernhard-42, 2025-07-10:
@MatthiasJ1 
I have uploaded 2.8.6 as preview to https://github.com/bernhard-42/vscode-ocp-cad-viewer/releases/tag/v2.8.6
This should fix this issue.
Maybe you could give it a try?
Thanks

MatthiasJ1, 2025-07-19:
Sorry I've been busy these last few weeks. I just checked it and now it never works. Previously, it would usually work but sometimes I would get an empty tab. Now I just get a completely blank area with no tab every time when reopening VSCode:

[image]

MatthiasJ1, 2025-07-19:
It works when I enabled `Ocp Cad Viewer › Advanced: Autostart` although only after selecting a python file. Would it not be simplest if when there is an `Ocp Cad Viewer` tab open, the viewer starts in that tab when VSCode starts up? Am I missing something?

MatthiasJ1, 2025-07-20:
There may be a miscommunication here. I am not suggesting that the viewer is opened automatically when you start VSCode, only that if it is already open it starts up. You can have multiple VSCode workspaces, and these save their window layout. For example, I have my 3d printing workspace. That workspace has a split pane layout, with the left pane containing tabs of code editors, and the right pane with a single OCP_Vscode tab. If I open a different workspace, for example not one for cad, I get just one pane with a code editor. If I open my cad workspace, that 2 pane layout already contains OCP_Vscode, except OCP_Vscode either decides it is needed and starts up properly, or sometimes decides it is not needed and you instead get a broken OCP_Vscode window. Then you get an empty pane in your workspace which needs to be closed and reopened manually. So I am not asking that OCP_Vscode is opened automatically when VScode starts, I am asking that if there is already a OCP_Vscode window open in the workspace, that window starts up regardless of whether it finds cad code.

bernhard-42, 2025-07-20:
> Then you get an empty pane in your workspace which needs to be closed and reopened manually

You actually don't need to close and open. Simple save your open cad python file or open a new cad python file and the viewer will kick into this pane (if autostart is on)

bernhard-42, 2025-07-20:
btw. I even try to close the viewer window on VS Code exit, but the time is not sufficient that VS Code provides for cleanups on exit.
Due to internal state there is no proper way to recover the last session. The viewer needs to be stared from scratch.
I thought the autostart mechanism is a sufficient approach ...

MatthiasJ1, 2025-07-20:
There is no need to recover the last session, just have the viewer startup if there is a viewer window open.

bernhard-42, 2025-08-05:
@MatthiasJ1 Restore of last session (i.e. restore from a VS Code persepctive, but actually a new start from the OCP viewer) is back. Hope it works for you

MatthiasJ1, 2025-08-08:
It appears to work well now, thank you!

closed 2025-08-08 by bernhard-42