bernhard-42/vscode-ocp-cad-viewer #181: OCP VS Code not found in newest version

cpeth, 2025-07-10:
Getting an error when opening VS Code stating the the library isn't found:

[image]

Just started happening with the latest update. Reverting to previous release fixes the issue.

cpeth, 2025-07-10:
Yes ocp_vscode is installed.

The python line returns `/usr/local/lib/python3.11/site-packages` however that is not where my ocp_vscode is installed so that node snipped wouldn't find it.

[image]

It's installed in ` /home/vscode/.local/lib/python3.11/site-packages`

When I downgraded to 2.7.1 and everything is fine, so for whatever reason older versions were able to locate the `ocp_vscode` package.

bernhard-42, 2025-07-10:
Thanks for the info! I did not consider that packages are install in user site

Older versions simply didn't check for ocp_vscode being installed. If you say "no" the new version should behave like the old one.

But I will fix in the current release, first need to find a reliable way to determine whether ocp_vscode is installed. (importing it takes too long, unfortunately)

bernhard-42, 2025-07-10:
Never mind, I found a better way to check whether ocp_vscode is installed (my library manager already knows what is installed, hence I only need to check)

bernhard-42, 2025-07-10:
fixed in prerelease 2.8.6 (see preview to https://github.com/bernhard-42/vscode-ocp-cad-viewer/releases/tag/v2.8.6)

cpeth, 2025-07-10:
Great, I will test out the prerelease shortly. Thanks for all your work on this extension!

cpeth, 2025-07-28:
Sorry for the delay. Yes it is fixed.