bernhard-42/vscode-ocp-cad-viewer #183: Empty config throws unhelpful exception

rmsc, 2025-07-15:
I used to have an empty config file, which I assumed would use the defaults.

After upgrading to v2.8, I started getting a weird JSON exception. It took me some time to figure out it was caused by said empty config.

I think this may be better handled by producing a more helpful error message:

```
Traceback (most recent call last):
  File "[redacted]/thumbwheel.py", line 42, in <module>
    show(wheel, render_joints=True)
    ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/show.py", line 514, in show
    return _show(*cad_objs, **none_filter(locals(), ["cad_objs"]))
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/show.py", line 568, in _show
    timeit = preset("timeit", timeit)
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/config.py", line 432, in preset
    return get_default(key) if value is None else value
           ~~~~~~~~~~~^^^^^
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/config.py", line 269, in get_default
    return get_defaults().get(key)
           ~~~~~~~~~~~~^^
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/config.py", line 274, in get_defaults
    result = dict(workspace_config())
                  ~~~~~~~~~~~~~~~~^^
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/config.py", line 480, in workspace_config
    port = get_port()
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/comms.py", line 126, in get_port
    find_and_set_port()
    ~~~~~~~~~~~~~~~~~^^
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/comms.py", line 329, in find_and_set_port
    port = find_port()
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/comms.py", line 291, in find_port
    ports = get_ports()
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/state.py", line 102, in get_ports
    return atomic_operation(callback)
  File "[redacted]/.venv/lib/python3.13/site-packages/ocp_vscode/state.py", line 64, in atomic_operation
    config = json.load(f)
  File "/usr/lib/python3.13/json/__init__.py", line 293, in load
    return loads(fp.read(),
        cls=cls, object_hook=object_hook,
        parse_float=parse_float, parse_int=parse_int,
        parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.13/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ~~~~~~~~~~~~~~~~~~~~~~~^^^
  File "/usr/lib/python3.13/json/decoder.py", line 345, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/json/decoder.py", line 363, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

rmsc, 2025-07-15:
Thanks!

The empty file I had was named `.ocpvscode`. The file created with `--create_configfile` was named `.ocpvscode_standalone`. Maybe this is why it crashed?

bernhard-42, 2025-07-15:
`.ocpvscode` is for the viewer in VS Code (port handling for several viewer instances)
`.ocpvscode_standalone` is for the standalone (substitute for the workspace setting in VS Code)

Do you use the standalone version or the VS Code integrated version?

bernhard-42, 2025-07-16:
It should be fixed in the 2.8.7 preview.
Some folks are testing it now and if all goes well I'll publish it as 2.8.8 to the VS Code marketplace

bernhard-42, 2025-07-28:
released 2.8.9

closed 2025-07-28 by bernhard-42