bernhard-42/vscode-ocp-cad-viewer #185: Request: Option for view buttons to only factor in visible objects

MatthiasJ1, 2025-07-20:
```py
plate = extrude(Rectangle(100, 100), -1)
loc = Pos(-50, -50) * extrude(Rectangle(1,1), 1)
```

Hide `plate`. All of these buttons are IMO now useless:

[image]

bernhard-42, 2025-07-20:
I don't agree that they are useless. They still do the right thing in 3d space.
If you want to get `loc` centered, use isolate (see help)

MatthiasJ1, 2025-07-20:
They are still behaving correctly within their defined parameters, however, the object of interest is a small blip in the corner of the screen.

In this scenario the `isolate` operation would work. If I instead had this:

```py
plate = extrude(Rectangle(100, 100), -1) # hide this in GUI
loc = Pos(-50, -50) * extrude(Rectangle(1,1), 1)
loc2 = Pos(-60, -50) * extrude(Rectangle(1,1), 1)
```

`isolate` would no longer work to view the shown objects.

bernhard-42, 2025-07-20:
understood, but I still don't see the need to make the logic more complex. It is very fast to just show the objects of interest, isn't it?
I already have quite a lot of modes and modifier keys, ..., and I don't think this make using the viewer easier

bernhard-42, 2025-07-20:
It wouldn't be the location buttons, btw., but we would need yet another modifier combination for "hide object and center the rest", in order to be consistent

MatthiasJ1, 2025-07-20:
> understood, but I still don't see the need to make the logic more complex. It is very fast to just show the objects of interest, isn't it?

One by one, yes, but when you have a complex assembly and you are working on a subset of that assembly comprised of multiple parts, navigation becomes challenging.

I agree additional modifier combos would not be great. I was suggesting a config (or kwarg) option that changes the behavior of the navigation buttons so they only factor in shown objects. Alternatively, the modifier keys could be on the navigation buttons since they don't have any.

> It wouldn't be the location buttons, btw., but we would need yet another modifier combination for "hide object and center the rest", in order to be consistent

What if you want to center the view on something you are already looking at? Having to show then unshow something else to accomplish this seems confusing.

MatthiasJ1, 2025-07-21:
Is that not what currently happens except it takes into account both shown and hidden objects?

bernhard-42, 2025-08-05:
@MatthiasJ1 Implemented in 2.9.0: Press shift when selecting the preview buttons in the toolbar

closed 2025-08-08 by bernhard-42