bernhard-42/vscode-ocp-cad-viewer #186: invalid literal for int() with base 10: 'version'

trayracing, 2025-07-24:
I get `invalid literal for int() with base 10: 'version'` when I call `show`. Someone on discord said they were able to work around it using `setport`, and that works. 
I see the release notes for 2.8.9 say:
>Add a robuts port in use detection to Windows

Maybe related?
I do not, AFAICT, have multiple OCP viewers open. This script worked in February.
I'm on OCP Viewer 2.8.9 and build123d 0.9.1.dev15+gf22f54a

Thanks

bernhard-42, 2025-07-25:
What is in ~/.ocpvscode?
Try to delete it, or replace content with {}

trayracing, 2025-07-25:
It had 
`{
  "version": 2,
  "services": {
    "3939": ""
  }
}`
I deleted it and reran. The issue still happened.

Then I discovered that in VSCode OCP backend gets its own terminal, and it saw ocp_viewer as 2.6.1. Pushing that forward to 2.8.9 resolved my problem. Last night I was looking in the VSCode debug terminal for ocp_viewer version. Thanks.

bernhard-42, 2025-07-25:
ah ok.
Glad it works again