bernhard-42/vscode-ocp-cad-viewer #189: Request to implement argument to set view from top, bottom, front, back, left, or right.

Technicus, 2025-08-08:
> Technicus — Yesterday at 16:27
> Is there a flag for OCP CAD Viewer that will set the view to top?  I am lookin int the [Standalone mode](https://github.com/bernhard-42/vscode-ocp-cad-viewer?tab=readme-ov-file#standalone-mode) documentation and in the definition, but not sure what it may be.
> Also, how do I set show axis at origin?

> bernhard — 12:47
> No, but I  can add it. 
> Could you maybe raise an issue in vcscode-ocp-cad-viewer, so that I don't forget it? 

> Jern — 12:50
> could input be a class property? That way a custom one can be defined but there are also pre-defined ones (Similar to build123d e.g. Plane.XZ)

> bernhard — 14:59
> It is a bit more complicated: The camera is controlled by position, quaternion (rotation), zoom, and target (where to look at).
> These 4 can already be set today, however, if they do not fit together, the viewer will typically look somewhere where no object is.
> The safest way of getting a view plane is to move the object as you want it, use status() to get the 4 variables mentioned above and then use these with show.
> The 7 preset views are fix defined and I can just expose them to the Python world

westurner, 2025-09-16:
> The safest way of getting a view plane is to move the object as you want it, use `status()` to get the 4 variables mentioned above and then use these with show.

- [ ] add a vscode setting for default view settings
- [ ] expose in the API the 7 preset fixed views in Python (but validate the enum strs first)
- [ ] expose in the API the the camera args: `position`, `quaternion` (rotation), `zoom`, and `target` (where to look at).

This might have to be another issue? @bernhard-42 

- [ ] preserve the view `status()` between rerenders with F5, so that the view stays the same when you render again

bernhard-42, 2025-09-16:
@westurner A `show` parameter `reset_camera` already exists that acts as you described and that can take three values:
- `Camera.RESET`: reset the position before drawing the next objects
- `Camera.CENTER`: Keep orientation (quaternion) before drawing the next objects, but center them to their bounding box center
- `Camera.KEEP`: Keep position, orientation and zoom before drawing the next objects

The next release will have 7 more parameters, `Camera.TOP` (top view), ..., `Camera.RIGHT` (right view), to orient the camera after drawing the options.

The more difficult part is the idea to provide a plane on the Python side and convert it to the correct position, quaternion, target and zoom for both perspective and orthographic camera.