bernhard-42/vscode-ocp-cad-viewer #196: Standalone mode appears to be broken.

quasipedia, 2025-09-24:
Running v. 2.9.0 on fedora linux, in a dedicated .venv.

I tried with both Firefox and Brave browsers (so: two different rendering engines), but **when trying to render anything the webpage at `http://127.0.0.1:3939/viewer` remains blank.**

Further down I have included the output from both the server and the script I tried to run, but I suspect the error may be with JS, as opening the console I got (firefox and brave respectively).

[image]

[image]

Initialising the server and registering the client.
```console
The viewer is running on http://127.0.0.1:3939/viewer

 * Serving Flask app 'ocp_vscode.standalone'
 * Debug mode: on
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:3939
Press CTRL+C to quit
 * Restarting with stat
Debug: 
Config: {'ambient_intensity': 1.0, 'angular_tolerance': 0.2, 'axes': False, 'axes0': False, 'black_edges': False, 'center_grid': False, 'collapse': 'R', 'control': 'trackball', 'debug': True, 'default_color': '#e8b024', 'default_edgecolor': '#808080', 'default_facecolor': 'Violet', 'default_opacity': 0.5, 'default_thickedgecolor': 'MediumOrchid', 'default_vertexcolor': 'MediumOrchid', 'deviation': 0.1, 'direct_intensity': 1.1, 'explode': False, 'grid': [False, False, False], 'metalness': 0.3, 'modifier_keys': {'ctrl': 'ctrlKey', 'meta': 'metaKey', 'shift': 'shiftKey'}, 'new_tree_behavior': True, 'glass': True, 'tools': True, 'pan_speed': 0.5, 'ortho': True, 'reset_camera': 'RESET', 'rotate_speed': 1.0, 'roughness': 0.65, 'theme': 'light', 'ticks': 10, 'transparent': False, 'tree_width': None, 'up': 'Z', 'zoom_speed': 0.5}

The viewer is running on http://127.0.0.1:3939/viewer

 * Debugger is active!
 * Debugger PIN: 726-302-924
127.0.0.1 - - [24/Sep/2025 23:29:11] "GET /viewer HTTP/1.1" 200 -
127.0.0.1 - - [24/Sep/2025 23:29:11] "GET /static/js/comms.js HTTP/1.1" 200 -
127.0.0.1 - - [24/Sep/2025 23:29:11] "GET /static/css/three-cad-viewer.css HTTP/1.1" 200 -
127.0.0.1 - - [24/Sep/2025 23:29:11] "GET /static/js/three-cad-viewer.esm.js HTTP/1.1" 200 -
127.0.0.1 - - [24/Sep/2025 23:29:11] "GET /static/js/logo.js HTTP/1.1" 200 -

Browser as viewer client registered

127.0.0.1 - - [24/Sep/2025 23:29:11] "GET /favicon.ico HTTP/1.1" 404 -
```

Running the example from the documentation:
```console
..cuments/3D objects/test [main|…9] ❯❯❯ uv run python example.py
Default port 3939 is open, using it
+
debug: handle_shapes (build123d Shape) (TopoDS_Compound) 
=> OcpObject name='Solid' kind=solid, ref=0, color=Color(232, 176, 36, 1.0), loc=((-0.5, -0.5, -0.5), (0.0, 0.0, 0.0, 1.0)), cache_id=fa7cb30028...

tessellation parameters:
 {'_splash': False, 'ambient_intensity': 1.0, 'angular_tolerance': 0.2, 'axes': False, 'axes0': False, 'black_edges': False, 'center_grid': False, 'clip_object_colors': False, 'clip_planes': True, 'collapse': 3, 'control': 'trackball', 'debug': True, 'default_color': '#e8b024', 'default_edgecolor': '#808080', 'default_facecolor': 'Violet', 'default_opacity': 0.5, 'default_thickedgecolor': 'MediumOrchid', 'default_vertexcolor': 'MediumOrchid', 'deviation': 0.1, 'direct_intensity': 1.1, 'explode': False, 'glass': True, 'grid': [False, False, False], 'helper_scale': 1.0, 'metalness': 0.3, 'modifier_keys': {'ctrl': 'ctrlKey', 'meta': 'metaKey', 'shift': 'shiftKey'}, 'new_tree_behavior': True, 'ortho': True, 'pan_speed': 0.5, 'render_edges': True, 'render_joints': False, 'render_mates': False, 'render_normals': False, 'reset_camera': 'keep', 'rotate_speed': 1.0, 'roughness': 0.65, 'show_parent': False, 'show_sketch_local': True, 'ticks': 10, 'timeit': False, 'tools': True, 'transparent': False, 'up': 'Z', 'zoom_speed': 0.5, 'clip_slider_0': 0.1, 'clip_slider_1': 0.15, 'clip_slider_2': 0.2, 'clip_normal_1': (0.12, -0.36, -0.92), 'clip_normal_2': (-0.58, 0.58, -0.58), 'clip_intersection': False, '_force_in_debug': False}
+
config:
 {'_splash': False, 'ambient_intensity': 1.0, 'angular_tolerance': 0.2, 'axes': False, 'axes0': False, 'black_edges': False, 'center_grid': False, 'clip_object_colors': False, 'clip_planes': True, 'collapse': 3, 'control': 'trackball', 'debug': True, 'default_color': '#e8b024', 'default_edgecolor': '#808080', 'default_facecolor': 'Violet', 'default_opacity': 0.5, 'default_thickedgecolor': 'MediumOrchid', 'default_vertexcolor': 'MediumOrchid', 'deviation': 0.1, 'direct_intensity': 1.1, 'explode': False, 'glass': True, 'grid': [False, False, False], 'helper_scale': 1.0, 'metalness': 0.3, 'modifier_keys': {'ctrl': 'ctrlKey', 'meta': 'metaKey', 'shift': 'shiftKey'}, 'new_tree_behavior': True, 'ortho': True, 'pan_speed': 0.5, 'render_edges': True, 'render_joints': False, 'render_mates': False, 'render_normals': False, 'reset_camera': 'keep', 'rotate_speed': 1.0, 'roughness': 0.65, 'show_parent': False, 'show_sketch_local': True, 'ticks': 10, 'timeit': False, 'tools': True, 'transparent': False, 'up': 'Z', 'zoom_speed': 0.5, 'clip_slider_0': 0.1, 'clip_slider_1': 0.15, 'clip_slider_2': 0.2, 'clip_normal_1': (0.12, -0.36, -0.92), 'clip_normal_2': (-0.58, 0.58, -0.58), 'clip_intersection': False, '_force_in_debug': False, 'normal_len': 0}

c
c

'reset_camera=True' is deprecated, use 'reset_camera=Camera.RESET' instead

c
+
+
c
+c
----++++++
```

In the server terminal there are several pages of debug output, but it is mostly a repetition/variation of:
```console
Debug: 
Config: {'ambient_intensity': 1.0, 'angular_tolerance': 0.2, 'axes': False, 'axes0': False, 'black_edges': False, 'center_grid': False, 'collapse': 'R', 'control': 'trackball', 'debug': True, 'default_color': '#e8b024', 'default_edgecolor': '#808080', 'default_facecolor': 'Violet', 'default_opacity': 0.5, 'default_thickedgecolor': 'MediumOrchid', 'default_vertexcolor': 'MediumOrchid', 'deviation': 0.1, 'direct_intensity': 1.1, 'explode': False, 'grid': [False, False, False], 'metalness': 0.3, 'modifier_keys': {'ctrl': 'ctrlKey', 'meta': 'metaKey', 'shift': 'shiftKey'}, 'new_tree_behavior': True, 'glass': True, 'tools': True, 'pan_speed': 0.5, 'ortho': True, 'reset_camera': 'RESET', 'rotate_speed': 1.0, 'roughness': 0.65, 'theme': 'light', 'ticks': 10, 'transparent': False, 'tree_width': None, 'up': 'Z', 'zoom_speed': 0.5}
127.0.0.1 - - [24/Sep/2025 23:30:21] "GET / HTTP/1.1" 200 -
Debug: Received config command
Debug: 
Config: {'ambient_intensity': 1.0, 'angular_tolerance': 0.2, 'axes': False, 'axes0': False, 'black_edges': False, 'center_grid': False, 'collapse': 'R', 'control': 'trackball', 'debug': True, 'default_color': '#e8b024', 'default_edgecolor': '#808080', 'default_facecolor': 'Violet', 'default_opacity': 0.5, 'default_thickedgecolor': 'MediumOrchid', 'default_vertexcolor': 'MediumOrchid', 'deviation': 0.1, 'direct_intensity': 1.1, 'explode': False, 'grid': [False, False, False], 'metalness': 0.3, 'modifier_keys': {'ctrl': 'ctrlKey', 'meta': 'metaKey', 'shift': 'shiftKey'}, 'new_tree_behavior': True, 'glass': True, 'tools': True, 'pan_speed': 0.5, 'ortho': True, 'reset_camera': 'RESET', 'rotate_speed': 1.0, 'roughness': 0.65, 'theme': 'light', 'ticks': 10, 'transparent': False, 'tree_width': None, 'up': 'Z', 'zoom_speed': 0.5, '_splash': True}
127.0.0.1 - - [24/Sep/2025 23:30:21] "GET / HTTP/1.1" 200 -
Debug: Received config command
Debug: 
Config: {'ambient_intensity': 1.0, 'angular_tolerance': 0.2, 'axes': False, 'axes0': False, 'black_edges': False, 'center_grid': False, 'collapse': 'R', 'control': 'trackball', 'debug': True, 'default_color': '#e8b024', 'default_edgecolor': '#808080', 'default_facecolor': 'Violet', 'default_opacity': 0.5, 'default_thickedgecolor': 'MediumOrchid', 'default_vertexcolor': 'MediumOrchid', 'deviation': 0.1, 'direct_intensity': 1.1, 'explode': False, 'grid': [False, False, False], 'metalness': 0.3, 'modifier_keys': {'ctrl': 'ctrlKey', 'meta': 'metaKey', 'shift': 'shiftKey'}, 'new_tree_behavior': True, 'glass': True, 'tools': True, 'pan_speed': 0.5, 'ortho': True, 'reset_camera': 'RESET', 'rotate_speed': 1.0, 'roughness': 0.65, 'theme': 'light', 'ticks': 10, 'transparent': False, 'tree_width': None, 'up': 'Z', 'zoom_speed': 0.5, '_splash': True}
127.0.0.1 - - [24/Sep/2025 23:30:21] "GET / HTTP/1.1" 200 -
Debug: Received status command
127.0.0.1 - - [24/Sep/2025 23:30:21] "GET / HTTP/1.1" 200 -
Debug: Received config command
```

bernhard-42, 2025-09-25:
It is a known bug, see https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/195, introduced in 2.9.0
Please use `python -m ocp_vscode --tree_width=240` for the time being

bernhard-42, 2025-11-05:
Fixed in 3.0.0